TonyRobotics/RoboWare #7: Support catkin build

ceesietopc, 2017-02-01:
Great project!

I am using catkin build instead of catkin_make, because I like to build just a couple of packages at a time when developing. When trying to use RoboWare, I get the following alert:

`The build space at '/home/companion/catkin_ws/build' was previously built by 'catkin build'. Please remove the build space or pick a different build space.`

It would be very nice if catkin build support was added. #

tonyrobotics, 2017-02-06:
@ceesietopc , thank you for your good advice. we'll improve it in upcoming releases.

mrath, 2017-08-30:
Update on this? This is the primary thing stopping me from using RWS...

beetleskin, 2017-09-04:
Maybe this `tasks.json` helps in the meanwhile:
```json
{
	"version": "0.1.0",
	"command": "bash",
	"args": [
		"-c"
	],
	"options": {
		"cwd": "${fileDirname}"
	},
	"isShellCommand": true,
	"showOutput": "always",
	"suppressTaskName": true,
	"tasks": [
		{
			"taskName": "Debug",
			"args": [
				"catkin build --this --no-deps --no-status --cmake-args '-DCMAKE_BUILD_TYPE=Debug'"
			],
			"problemMatcher": {
				"owner": "cpp",
				"fileLocation": [
					"absolute"
				],
				"pattern": {
					"regexp": "^(.*):(\\d+):(\\d+):\\s+(warning|error):\\s+(.*)$",
					"file": 1,
					"line": 2,
					"column": 3,
					"severity": 4,
					"message": 5
				}
			}
		},
		{
			"taskName": "Release",
			"args": [
				"catkin build --this --no-deps --no-status --cmake-args '-DCMAKE_BUILD_TYPE=Release'"
			],
			"problemMatcher": {
				"owner": "cpp",
				"fileLocation": [
					"absolute"
				],
				"pattern": {
					"regexp": "^(.*):(\\d+):(\\d+):\\s+(warning|error):\\s+(.*)$",
					"file": 1,
					"line": 2,
					"column": 3,
					"severity": 4,
					"message": 5
				}
			},
			"isBuildCommand": true
		}
	]
}
``` 

You can easily add your own catkin build chain as another task.

wangbinhai, 2017-09-08:
@mrath not yet. working on it

tonyrobotics, 2017-09-15:
Catkin build support will be added at the end of October.

crashtest84, 2017-11-22:
Looks like it didn't make it? I just updated to 1.0.1 it doesn't appear to like catkin build spaces

skohlbr, 2017-11-23:
Same problem here. I think a lot of  people working with large projects switched to catkin tools, so supporting this should be a priority.

Walmpie, 2018-02-21:
What's the status of this issue? I'm unable to import a 'catkin build' project with version 1.1.0.

tonyrobotics, 2018-02-21:
@Walmpie , both 'catkin_make' and 'catkin build' are supported now, but there should still be a CMakeLists.txt in the **src** folder to tell RoboWare it's a ROS project. You could run ‘catkin_init_workspace’ in **src** to generate one.

Walmpie, 2018-02-21:
@tonyrobotics Thanks for the quick answer. The workspace has been initialized by the catkin_tools command 'catkin init' so there is no top-level CMakeLists.txt. I could initialize it with the command you propose. However this doesn't belong to catkin_tools right?

tonyrobotics, 2018-02-21:
@Walmpie You're right, the job of 'catkin_init_workspace' is just to generate the CMakeList.txt file (actuallly a symbol link) in the **src** folder. 'catkin init' wouldn't do this job, so RoboWare regards it as an invalid workspace. Everything will be OK as soon as there's a file (or symbol link) named **CMakeLists.txt** in the **src** folder.

Walmpie, 2018-02-22:
@tonyrobotics, this works. Thanks for the help :+1: