TonyRobotics/RoboWare #9: Please install [clang](http://clang.llvm.org/) or check configuration `clang.executable`

nickhuan, 2017-03-13:
This error message is shown every time when I click on my cpp source file. And I have already installed clang3.8 using apt-get, but the problem is still there.

tonyrobotics, 2017-03-13:
@nickhuan , which version of roboware are you using? When execute "clang -v" in terminal, what is the output?

nickhuan, 2017-03-13:
The version I'm using is 0.4.2:
"Version 0.4.2
Commit 71c238851d88c4aaa2853cc7768678ead4b1a96c
Date 2017-02-28T01:50:02.302Z
Shell 1.4.6
Renderer 53.0.2785.143
Node 6.5.0" in About

I get following when I execute "clang -v":
The program 'clang' can be found in the following packages:
 * clang-3.3
 * clang-3.4
 * clang-3.5
Try: sudo apt-get install <selected package>

tonyrobotics, 2017-03-13:
@nickhuan , clang is not installed properly, try "sudo apt-get install clang" to install clang. 
After installation, execute "clang -v'', the following outputs indicate that clang is successfully installed:

clang version 3.9.0 (tags/RELEASE_390/final)
Target: x86_64-unknown-linux-gnu
...

nickhuan, 2017-03-13:
The system requires me to install clang-3.4 and I think it works fine now.Thanks