TonyRobotics/RoboWare #11: debug not work for some ROS project

jacknlliu, 2017-03-29:
If clone a ROS project [universal_robot](https://github.com/ros-industrial/universal_robot), and copy it to a `catkin_ws/src`, then get a file structure like:
```
catkin_ws
└── src
    ├── CMakeLists.txt
    └── universal_robot
```
Open this workspace with roboware, and follow the `debug` instruction, but I can't get the ros nodes and executable files in debug mode.

tonyrobotics, 2017-04-10:
`universal_robot` is a git repo name, not a ros package, your file structure should like this:
```
catkin_ws
    └── src
        ├── CMakeLists.txt
        ├── universal_robot
        ├── ur10_moveit_config
        ├── ur3_moveit_config
        ├── ur5_moveit_config
        ├── ur_bringup
        ├── ur_description
        ├── ur_driver
        ├── ur_gazebo
        ├── ur_kinematics
        └── ur_msgs
```

jacknlliu, 2017-04-13:
Thank you very much for your reply.

closed 2017-04-13 by jacknlliu