TonyRobotics/RoboWare #13: GIT improvements

TobiMiller, 2017-04-10:
Hey,

is it possible to improve the implementation of git? i dont want the devel or build dir in my git repo. please insert a option to set the git repo dir for the ros workspace.

thx in advance

beetleskin, 2017-09-04:
Just edit your `.gitignore` file accordingly ..?

tulayang, 2017-09-08:
@TobiMiller 

You can add a `.gitignore` file in your project, then edit it as:

```sh
*.~
devel
build
```

SammyRamone, 2018-03-05:
The usage of Git in RoboWare should really be improved. It is not possible to have packages in the src folder which are gits themselves. They are not recognized. Furthermore, I don't think that it is the usual approach to have a Git of your workspace, but rather several Gits of your code.
I tried VS Code with the ROS extension and there it is possible to specify multiple Gits  and manage them nicely. Still Roboware has more ROS related features, but the missing Git support is really a downside.

bionade24, 2019-05-15:
Why is it so hard to find the git Repos as it works in VSCode?