TonyRobotics/RoboWare #14: RMP for Fedora

MalcolmMielle, 2017-04-24:
Hello,

I'm not 100% sure I should post this here so if I shouldn't please let me know and I'll move it. I use rox on Fedora and I'll be interested in using RoboWare, could we have the source available ?

wangbinhai, 2017-05-05:
Thank you for your interest. We haven't made RoboWare open source by now but if you really need it we will try to pack a version for Fedora. We didn't try it in Fedora before so we cannot guarantee it works.

Regards
Tony

jacknlliu, 2017-09-08:
@MalcolmMielle  you may need a docker version. You can consider my ros docker image [jacknlliu/ros](https://hub.docker.com/r/jacknlliu/ros/) which integrated with roboware. I think this will significantly improve your user experience about ros on Fedora. 

Best regards!