TonyRobotics/RoboWare #15: Is source code available?

wolfd, 2017-04-27:
I'm trying out this IDE, and would love to know if there's source code available.

Best,
wolfd

wangbinhai, 2017-05-05:
Sorry but currently we haven't made it open source.

Regards
Tony

bkinman, 2018-02-11:
When will it be open source? I was told months ago that it would be made open?

beetleskin, 2018-02-12:
Well, although they don't upload the "source code" here, you can just grab a release and have a look into the package content. Everything lies there as plain source. The release is just a bundled version of vscode, with addons, scripts and settings. Nothing is compiled or obfuscated, as far as I remember.

SammyRamone, 2018-03-05:
@beetleskin seems to be right
Why not make it "really" open source by putting the files directly into Github, not as .deb file. Then others could make pull request or find bugs easier.

tony-robotics, 2018-06-20:
Roboware Studio is open sourced now! Its source code is available at https://github.com/tonyrobotics/roboware-studio.

closed 2018-11-26 by wolfd