TonyRobotics/RoboWare #22: Add catkin_tools task profile

malcolmwhat, 2017-07-17:
_Since the main draw of this is easy setup:_
This isn't difficult to do manually, but it would be great if there was either

1. Add a starter `tasks.json` for a catkin_tools build system
2. Integrate a set of default catkin_tools oriented build and release steps and make the build tool select-able by the user

wangbinhai, 2017-07-20:
Thanks Malcolm, would put it in the list.

mrath, 2017-08-30:
@malcolmwhat - did you in fact get RWS to open a catkin_tools workspace? 
Adding cyclic reference to #28...

wangbinhai, 2017-08-30:
We are working on it

mrath, 2017-08-30:
Awesome!
(Not to bother you more, but I think you should release this as open source so people can contribute. It has a lot of potential for the community, since there is currently no good starter IDE, and this is the best candidate.)

slashie, 2017-08-31:
I agree with @mrath, would love to contribute to the project.

beetleskin, 2017-09-05:
Duplicate of #7, also see #7#issuecomment-326974922