TonyRobotics/RoboWare #28: Unable to open some workspaces

naveedhd, 2017-08-10:
Some of my workspaces don't open in RoboWare Studio resulting in:

```
Path is not ROS workspace
The path '/home/me/ws' does not seem to be an effective ROS Workspace on disk.
```

This workspace is built using `cakin build` rather than `catkin_make`.

How does the roboware studio distinguish a catkin workspace?

slashie, 2017-08-29:
just installed, having same issue, using `catkin build`

mrath, 2017-08-30:
Same issue here. Maybe Malcolm in #22 got it working?

beetleskin, 2017-09-05:
Right now, only the "raw" catkin is supported, not the catkin_tools extension. You can resolve this issue by by copying the CMakeLists.txt from any workspace created with `catkin_init_workspace` (ist just a symbolic link anyway)

@tonyrobotics there seems to be a lot of confusion about your support of _catkin tools_ along with a lot of duplicate issues. Adding some info to the Readme.md might help ;)

tulayang, 2017-09-08:
As @beetleskin said, a **CMakeLists.txt** generated by `catkin_init_workspace` is necessary, this file should be in **src** directory.

roboticsai, 2018-03-19:
ok, after i created a CMakeLIsts.txt file inside src folder of my ros workspace i.e by initializing that this folder is made for ros, robware is smoothely accepting any catkin workspace .