TonyRobotics/RoboWare #31: 1. Move the cursor to class/function/variable name or head file 2. Go to Definition(F12) 3. Pop the tip "No definition found for ..."

xiankaichen, 2017-09-01:
- VSCode Version: RoboWare Studio 0.7.2 (124306487bf93b6675696cabf94bcf8580f35b84, 2017-08-24T02:11:58.349Z)
- OS Version: Linux x64 4.4.0-78-generic
- Extensions: none
---

Steps to Reproduce:

1. Move the cursor to class/function/variable name or head file
2. Go to Definition(F12)
3. Pop the tip "No definition found for ..."

beetleskin, 2017-09-04:
I wonder why ycmd was chosen over the default cpp extension ..

tonyrobotics, 2017-09-05:
> Can confirm this, and I wonder why ycmd was chosen over the default cpp extension ..

Default cpp extension doesn't work well for code auto-completion,  ycmd is doing better.

beetleskin, 2017-09-05:
Okay, is there a trick to it? Or can you, too, confirm the ""No definition found for ..." behaviour? Within my project, it seems to work for some pieces of code and fail for others. Did I miss something in the setup?

JWSLAM, 2018-01-26:
also come across the same problem， is there any solution？

benbenji, 2018-01-28:
I aslo have this problem is there some solution？ Is this some problem of ycmd？

benbenji, 2018-01-29:
I have found the solution.I download the ycmd from github and build all the things by myself.Then copy the fold like this
cp ~/ycmd /usr/share/roboware-studio/resources/app/extensions/RichardHe.you-complete-me-1.0.36/ycmd
and then copy ycm_extra_conf.py and extra_conf.py from origin ycmd to new fold.
The jump function can run normally .But there is still something confusing in auto completion.It often makes up some unrelated options.But I am satisfy the result now.Hope that is help for you
I hope roboware can get better and better！It is really useful and necessary. @xiankaichen @tonyrobotics