TonyRobotics/RoboWare #42: 我用sftp为什么会“磁盘上路径“/run/user/1000/gvfs/sftp:host=192.168.1.177/home/flx/catkin_ware”似乎不是一个有效的 ROS 工作区”这样的错误呢

kevinchow1993, 2018-01-09:
- VSCode Version: RoboWare Studio 1.1.0 (9a9971530fa5b18a24ec3b1b6215edd868669033, 2017-12-27T00:37:19.062Z)
- OS Version: Linux x64 4.4.0-104-generic
- Extensions: none
---

Steps to Reproduce:

1.
2.

kevinchow1993, 2018-01-09:
因为尝试远程调试始终失败，所以使用了这个方法，用sftp打开远程资源管理器
[image]
[image]
再在roboware studio中创建catkin_ware工作区

beetleskin, 2018-01-09:
When you stated your issue in English, much more developers could help you out.

tonyrobotics, 2018-01-10:
RoboWare does not support creating or opening ROS workspace via SFTP protocol. Remote debugging is implemented via SSH protocol. 
For more information of remote debugging please refer to the user manual. 
Just as @beetleskin said, we need you to provide more specific error messages of remote debugging.