TonyRobotics/RoboWare #43: Roboware can't find C++ STL libraries.

DogggLee, 2018-01-22:
My platform  is Ubuntu 16.04 + ROS kinetic 
Before using Roboware, I use Qt + qtc_ros_plugin to edit my ROS projects and it works fine.

But there is some problem with include files when I use Roboware to edit my code. 
It can only find the ros-relative .h files or some libraries following by '.h' 
`#include <ros/ros.h>` 
`#include <sensor_msgs/Image.h> `
`#include <math.h>`
`#include <sys/types.h>` 
It can't find the C++ STL libraries(When I press 'F12' it says that 'No definition found for string') and auto completion also doesn't work on these variables (string, vector, etc.)
`#include <string>`
`#include <vector>`
`#include <fstream>`
But I have no problem for building. So is there somewhere to set the include directories or should I do some other configuration? 

Sorry for my poor English :(

tonyrobotics, 2018-01-22:
Hi Doggglee, we need you to provide more detailed information. Select "View"->"Output", there will be a top-down menu in the upper right of the "Output" window, choose "ycm-language-server" option and see what's the output?

DogggLee, 2018-01-22:
@tonyrobotics 
It turns out two warnings:
`(node:8362) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): SyntaxError: Unexpected end of JSON input`

`(node:8362) DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.`