TonyRobotics/RoboWare #45: Automatic completion

benbenji, 2018-01-28:
- VSCode Version: RoboWare Studio 1.1.0 (9a9971530fa5b18a24ec3b1b6215edd868669033, 2017-12-27T00:37:19.062Z)
- OS Version: Linux x64 4.4.0-111-generic
- Extensions: none
---

Steps to Reproduce:

1.when I want to jump some variable or function but many times it response "no definition for .."
2.Then I change the ycmd.showdebug to true,sometimes they response "internal error no translation unit"
Could you please tell me the solution? Think you very much! I am waiting for your reply.

beetleskin, 2018-01-28:
Why are you opening this issue another time? You already saw the other one (#31) ..

benbenji, 2018-01-29:
I have found the solution.I download the ycmd from github and build all the things by myself.Then copy the fold like this
cp ~/ycmd /usr/share/roboware-studio/resources/app/extensions/RichardHe.you-complete-me-1.0.36/ycmd
and then copy ycm_extra_conf.py and extra_conf.py from origin ycmd to new fold.
The jump function can run normally .But there is still something confusing in auto completion.It often makes up some unrelated options.But I am satisfy the result now.
I hope roboware can get better and better！It is really useful and necessary.

closed 2018-02-13 by benbenji