TonyRobotics/RoboWare #46: Remote Arguments not configured when I try to build my workspace

carlos-tubio, 2018-02-16:
Hello,

I am using  Version 1.1.0 Commit 9a9971530fa5b18a24ec3b1b6215edd868669033 version of Roboware.
When I try to compile my workspace using build option Remote Arguments not configured appears as a output in the output terminal of roboware.

I have no problem to compile my workspace from terminal  (catkin_make).
As a clue I have several workspaces in my PC.

Regards

tonyrobotics, 2018-02-17:
It should work. Remote configuration is a little bit complicated. Please refer to the manual [http://www.roboware.me/#/Download](url) for more details.

asimay, 2018-02-26:
[image]

same with me!!  I put  a git clone repo, and when building, this problem occurs!

tonyrobotics, 2018-02-26:
@asimay , I guess you are working on a local project, not a remote one. So you should choose the right option as follows (not remote):
[image]

asimay, 2018-02-26:
[image]

hi, dear, yes, it can work now. as I first download it and use, I didn't change the default setting. maybe you can fix it.  after build, I click run, and the windows pop error info, I checked and find it was env source issue, maybe you can also fix this bug, when run launch , no need to load bashrc to change, you can directly source env and then load launch.

Thanks!

tonyrobotics, 2018-02-26:
@asimay Thanks! That's a very good suggestion, we'll take this feature into consideration for future versions.

HARISYOUSAFZAI, 2019-05-27:
[image]
hi there! i am trying to build this project but its not working. in output it says "bash: catkin: command not found"
when i try to build this code through terminal it has been executed properly without any issue.
one more thing why cant i see many options in explorer as i have seen here in pictures.
im new to ros, ubantu, and roboware.  :(

HARISYOUSAFZAI, 2019-05-27:
> [image]
> hi there! i am trying to build this project but its not working. in output it says "bash: catkin: command not found"
> when i try to build this code through terminal it has been executed [...]

this problem is solved by editing in preference>setting>ROS
:)
this page is lucky for me :dancer: