TonyRobotics/RoboWare #52: catkin_make not working when called from Roboware

alberto-tellaeche, 2018-03-20:
Hi all;
I have installed the latest Roboware version available (1.1.0), as a new IDE for ROS.
Seems to detect my workspace and everything works just fine, except:

1.- When I try to build my packages using ROS->build menu command, I get the error "bash: catkin_make: command not found" . From terminal, it works ok. Any solution?

2.- The command "Debug-> Start debugging" gives the following error: " Failed to load MI Debugger: "/home/alberto/kinetic_ws/src/fegemu_demostrator/src/fegemu_demostrator.cpp": not in executable format: File format not recognized (from file-exec-and-symbols "/home/alberto/kinetic_ws/src/fegemu_demostrator/src/fegemu_demostrator.cpp")"

Can anyone help me with these two issues?
Thank you in advance, 
Alberto

fredrich-leo, 2018-10-14:
I'm stuck in the same problem, hope you guys can figure it out later. Enough of it V_V

Combinacijus, 2020-07-07:
For some reason terminal which runs catkin_make haven't sourced ros setup file. It's strange because it was working before but after reinstalling Ubuntu it broke.

**Workaround for _bash: catkin_make: command not found_**
1. In Roboware Studio **open any ROS workspace**.
2. At the top left corner to the right of drop down menu |Debug &nbsp; &nbsp; \\/| **click a gear icon**. It will open tasks.json file.
3. **In tasks.json file find "catkin_make"**. For debug configuration it will look like:
```
"taskName": "Debug",
"args": [
	"catkin_make -C ${workspaceRoot} -DCMAKE_BUILD_TYPE=Debug"
],
```
4. **Prefix catkin_make with 'source <ros_installation_setup.bash_path>; '** and save the file. For me it looks like (note that I use ros melodic and it's installed in /opt directory):
`"source /opt/ros/melodic/setup.bash; catkin_make -C ${workspaceRoot} -DCMAKE_BUILD_TYPE=Debug"`
5. **Repeat 3rd and 4th steps for other build configurations** you want (Release, Debug (isolated), etc.).
6. **Press build icon** and it should work.