TonyRobotics/RoboWare #54: Failed to load MI Debugger

roboticsai, 2018-03-20:
- VSCode Version: RoboWare Studio 1.1.0 (9a9971530fa5b18a24ec3b1b6215edd868669033, 2017-12-27T00:37:19.062Z)
- OS Version: Linux x64 4.13.0-37-generic
- Extensions: none
---

> Failed to load MI Debugger: "/home/robot/orocos_gazebo_ws/.vscode/tasks.json": not in executable format: File format not recognized (from file-exec-and-symbols "/home/robot/orocos_gazebo_ws/.vscode/tasks.json")

Steps to Reproduce:

1.create new ros workspace
2.clone https://github.com/jhu-lcsr/rtt_gazebo.git into workspace
3. run release isolated then debug isolate.

danneboom, 2018-05-13:
Same here for .cpp file. Are there any possible reasons, fix, workarounds?

paper2Fox, 2019-02-28:
Same.Have u found a solution for the problem? 
> Failed to load MI Debugger: "/home/jcl/TX/src/xrobot_base/launch/xrobot_odom_encoder.launch": not in executable format: File format not recognized (from file-exec-and-symbols "/home/jcl/TX/src/xrobot_base/launch/xrobot_odom_encoder.launch")

shanpenghui, 2019-04-02:
> Same.Have u found a solution for the problem?
> 
> > Failed to load MI Debugger: "/home/jcl/TX/src/xrobot_base/launch/xrobot_odom_encoder.launch": not in executable format: File format not recognized (from file-exec-and-symbols "/home/jcl/TX/src/xrobot_base/launch/xrobot_odom_encoder.launch")

hello, I also have this problem, had you solved it?

danneboom, 2019-04-02:
Not a solution, but I now use qtcreator with ros plugin instead.

paper2Fox, 2019-04-18:
> hello, I also have this problem, had you solved it?

No,can't find a solution.I use clion as a instead

Nronaldo, 2020-05-15:
Here may be a solution. 
You should add a breakpoint in a source file(.cpp), but you should invoke debug for a node.  
Look at your source manager(资源管理器) and select the node file and then debug. 
I am a newbie to ros and I am not sure this way can solve all questions.