TonyRobotics/RoboWare #62: text display issue

chaolmu, 2018-07-10:
- VSCode Version: RoboWare Studio 1.1.0 (9a9971530fa5b18a24ec3b1b6215edd868669033, 2017-12-27T00:37:19.062Z)
- OS Version: Linux x64 4.15.0-24-generic
- Extensions: none
---

Text letters in menu and source code are displayed improperly in Ubuntu 18.04 and other ubuntu based OS.
- menu: overlap
- source code: wrong letters.

KrzysztofDabek, 2019-03-14:
I have the same problem. Some of the test is badly scaled. Happens in random places 

I am using VirtualBox and Ubuntu 18.04, RoboWare Studio 1.2.0
[image]

MCFurry, 2019-04-08:
Same issue here with RoboWare Studio:
Version 1.2.0
Commit dcd7482df29ff9357bee10e008adef2fc475ac43
Date 2018-04-26T02:24:41.213Z
Shell 1.6.6
Renderer 56.0.2924.87
Node 7.4.0
on Ubuntu 18.04

Looks exactly like @KrzysztofDabek's post.