TonyRobotics/RoboWare #64: build error

TifferPelode, 2018-08-03:
Hello,
    When I use RoboWare View to build a new project,I got this:

> electron@1.8.7 postinstall /home/tiffer/viewer_test/build_test_tiffer pelode_1533266752307/node_modules/electron
> node install.js
sh: 1: node: not found
npm WARN test@0.0.1 No description
npm WARN test@0.0.1 No repository field.
npm ERR! Linux 4.15.0-29-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! node v4.2.6
npm ERR! npm  v3.5.2
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! electron@1.8.7 postinstall: `node install.js`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the electron@1.8.7 postinstall script 'node install.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the electron package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node install.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs electron
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls electron
npm ERR! There is likely additional logging output above.
npm ERR! Please include the following file with any support request:
npm ERR!     /home/tiffer/viewer_test/build_test_tiffer pelode_1533266752307/npm-debug.log

and this is my log file
[npm-debug.log](https://github.com/TonyRobotics/RoboWare/files/2255912/npm-debug.log)

I wanna to know what's wrong with this and how can I solve it?

uli2k, 2018-08-23:
You may need to install nodejs. I'm really sorry for the late reply. Please inform me if this cannot be solved.

huoxingwen, 2019-08-08:
you need install nodejs, and run sh build.sh need root